ubilab-ws21/puzzle-5 #3: Evaluate LCD library

archerindigo, 2021-12-06:
Evaluate LCD library to see how to control image displays.

May break this issue into smaller issues.

archerindigo, 2021-12-17:
@wjb321 May you set some smaller goals regarding the development of the HoloCube software and update the progress from time-to-time?

wjb321, 2021-12-17:
> 
I will set several subgoals and put them below.

archerindigo, 2021-12-17:
You should create new issues when you set your sub-goals. Better fit a week of workload for each task.

wjb321, 2021-12-18:
there are some basic tasks that I did in local disk about LCD（lvgl） and recently I will upload the local notes and update the tasks below:

1. [ ]vscode environment and esp-idf environment setup
2. [ ]esp32 examples tests and peripheral test
3. [ ]lvgl display interface and GUI library porting
4. [ ]lvgl animation test and coding 
5. [ ]2048 game + BLE control

wjb321, 2021-12-18:
1. [ ] vscode environment and esp-idf environment 
**a** first time i tried on mac by using [vscode_mac](https://code.visualstudio.com/download), but can not upload to board, it may be caused by driver problems or Type-c port  problem. I can build but can not upload
**b**  vscode environment here is my reference link [vscode_for_win (https://blog.csdn.net/yanwennian/article/details/103372880), this works fine.
**c** esp-idf, there were many problems. by checking the [manual](https://docs.espressif.com/projects/esp-idf/en/v4.3.1/esp32/get-started/index.html), i stalled the esp-idf tools installer(esp-idf power shell, cmd and eclipse) and then all the configurations for changing the layout and size should configure by some commands, somehow not so convenient for me. as an alternative, i found the the support plug-ins in the vscode, then i searched them and installed. then the problems happened with the environment variables conflicts, vscode esp-idf can not find some paths and files even i added in the path josn. but finally worked when reinstall the plugs-ins and make a fixed environment variables(can not share with original variables)
**d** tested some given examples in the Soc

wjb321, 2021-12-19:
**e,** sometimes building a esp-idf project in vscode, the menuconfig will stop in initial values state forever :(, for solving this issue, it took quite a long time to solve, the final solutions were either make sure  the repository esp-idf is the same root directory in the file or need configure the environment path.
**f,** the project opened in vscode can not be second level dir, it will make build file in the same level with the project, then the menuconfig will stop at loading initial variables forever again.
**g,** when esp-idf is in the same file, the example sub-file can be deleted, somehow can make the building time shorter
**h,** esp-idf building steps [esp-idf-soc](https://docs.espressif.com/projects/esp-idf/en/v4.3.1/esp32/get-started/index.html#step-1-install-prerequisites)

wjb321, 2021-12-19:
3. [ ] lvgl display interface, codeblocks simulator testing and GUI library porting

wjb321, 2021-12-20:
**A, **after clone all the files mentioned above then start the journey. in driver file(show above), there are mainly 2 parts, one is the tft driver, another one is touch driver. here I use st7989 ips lcd and no touch function, so I remove the touch driver.     
**B, **Code porting, there are main three files need to be modified in order to drive different LCD.
         > User_Setup_Select.h,   > copy lv_conf_template.h and rename it as lv_conf.h   > user_setup.h, these files according different requirements, need to be modified. reference sorry for Chinese [GUI guide](https://www.cxybb.com/article/qq_45516773/119673328)       
**C, **Now come to GUI design, according to different components, build different labels and animations. here is the simulator that I used are Visual studio and code blocks, and after testing working fine, then I porting to LVGL projects and run it.

wjb321, 2021-12-23:
6. [ ] here add one more thing: pictures loading

wjb321, 2022-01-03:
for picture loading there are mainly 2 options:

1. convert the pic into c online and put into flash(files)and read the c array of the given pictures
2. convert the pic into binary format and put them into SD card and read the pic from sd card. for reading from sd card:  
     - [ ] configure the sd driver: sd_card.cpp: where contains basic functions: init(), listDir, readFileLines, readBinFromSd, writeBintoSd.....
     - [ ] configure SPI(here sd card use spi mode), there are 3 different spi buses: FSPI, HSPI and VSPI, and due to my test board is already fixed module, so I redefine the pins to: ss: 0, clk:21, miso:22, mosi:19
     - [ ] configure FATFS of lvgl,  the driver file named: lv_port_fatfs.h/c are the basic for lvgl get some files(png, gif, bin and txt) from sd card.

wjb321, 2022-01-03:
now can load the pic from flash and read txt file from sd card, but for reading some gifs from sd card or flash still can not work. 

- [ ] animate the image as the start UI
- [ ] add hint for next puzzle 
- [ ] mpu combine and test

wjb321, 2022-01-23:
done with the drivers of gif files, it mainly has these configurations:
gif drivers needed:

> lv_gif.h/c   same like img with these 
   lv_gif_create()
   lv_gif_set_src()
   lv_gif_restart()

> gifdec.c/h  decode the gif(open gif data or file name and render frame)

TFT_eSPI  drivers needed:

> User_Setup.h : uncomment the needed ips drives #define and relative ips pins 
> User_Setup_Select.h: uncomment the needed ips setups

I tried get some online gif and cropped gif into needed size and convert gif online into raw  and .c array(flash read), but it is not working(only blinking) I think it can be the online cropping tool has some influence. and tried another converted gif, it works fine. 
I would like to change the interface of the game into gif