ubilab-ws21/puzzle-5 #5: Get to know the decorations to be done

archerindigo, 2021-12-06:
How should the decorations be done (especially the electric box)? Is environment team is responsible to help?

lathifsharieff, 2021-12-06:
I dont think they would be incharge of the led stripe inside the safety box maybe we need to configure the lighting and decorations accordingly.

archerindigo, 2021-12-10:
The box is fine, maybe we just need to add some signs to show it is a charging station and it can be unlocked by playing with the HoloCubic

The electric box is a bigger concern. How to turn a cupboard into a electric box and highlight the battery platform would be a challenge.

archerindigo, 2021-12-17:
Update: The NFC reader can be placed anywhere in the control room, not necessary to be in the cupboard. Böttcher suggested that the decoration can be decided later.

archerindigo, 2022-02-18:
At last, no decorations regarding the environment had to be done. The electric box's battery dock is placed on the table next to the server room's door. The charger is placed inside the safe. A hole is drilled at the right side of the safe to add an additional USB cable for the HoloCube.

closed 2022-02-18 by archerindigo