ubilab-ws21/puzzle-5 #28: 2048 game development on HoloCube

archerindigo, 2021-12-17:


wjb321, 2021-12-18:
1. [ ] 2048 game programming and test on visual studio
2. [ ] LVGL Animation analysis and 2048 realization 
3. [ ] IMU driver and lvgl combine -> game control
4. [ ] wechat app control 2048 by using BLE

wjb321, 2021-12-19:
1. [ ]2048 game first developed on VS, and using the library Easyx for images UI. there are some basic logics of the game:
**A,** initial map, img and imgindex respectively are map matrix, class img for storing the selected num.png, array for the numbers with same order like class img     
**B,**loadimag(easyx lib), just load the imags that you selected       
**C,**drawmap, base on the size  design the map, main func putimage()   
**D,**generate numbers, because after moving one step there will be new number generated      
**E,**basic moving directions(L,R,U,D) functions
**F,**test which key is pressed? keyPress
**G,**new hint imag generates when reach set goal

wjb321, 2021-12-19:
[image]
[image]
[image]

wjb321, 2022-01-03:
realizing the BLE control 2048 game on esp32. briefly designed the UI, there are some issue need to be done further

- add gif or img in the start page and by sending command to move the start page and move to the game page
- after finishing the game, there will be a UI showing the hint for next puzzle

wjb321, 2022-01-03:
[image]

wjb321, 2022-01-03:
[image]

wjb321, 2022-01-04:
- [ ] for the game, it would be better to make some interface and application package then the animation of images can work well, so next step should be this.
- [ ] by calling the function lv_scr_load_anim() can realize the screen switch[how to switch the UI](https://blog.csdn.net/yunjie167/article/details/105488356)

wjb321, 2022-01-09:
realize the interface of BLE control, can remote control game and shift game picture

wjb321, 2022-01-09:
- [ ] after reaching needed target, there will be a hint or png for next puzzel
- [ ] try GIF/Animation for the app interface
- [ ] try read png from sd card and make mpu works on the cube

wjb321, 2022-01-24:
the upper tasks are done, only the mpu and gif have been left
for gif, the lib is v8 while the project for game is based on v7, so the driver porting will take some time, i will update it later