ubilab-ws21/puzzle-5 #29: Bluetooth connection between HoloCube and charger box's MCU

archerindigo, 2021-12-17:
Apart from showing the box's password on the HoloCube, we may alsounlock the box directly through Bluetooth.

wjb321, 2021-12-18:
1. [ ] Know the theory of bluetooth and BLE
2. [ ] test different application scenarios of BLE 
3. [ ] holocubic as a client and charger box's mcu as a server v.v.
4. [ ] BLE further application

wjb321, 2022-01-03:
there are two types of bluetooth: bluetooth and BLE, because I used iPhone and Mac, so bluetooth normally can not support well for iOS. so BLE works well.

- [ ] now realized phone as the client and use BLE to control holocubic to play the game

wjb321, 2022-01-14:
now BLE works fine and can control the game. for the commands or information sending to mcu in charger box now change from bluetooth way to mqtt way.  so for the task mentioned above will do after mqtt finished.

closed 2022-01-28 by wjb321