ubilab-ws21/puzzle-5 #42: MQTT implement  and command sending

wjb321, 2022-01-14:
- [ ] theory and configuration of mqtt in esp32
- [ ] commands sending when game playing 
- [ ] extend some other application by mqtt

wjb321, 2022-01-14:
[reference of mqtt(chinese)](http://www.1zlab.com/wiki/micropython-esp32/mqtt/), basic info how mqtt works in esp32 and the way how to build the iot in esp32, which is the basic structure named esp-idf.

wjb321, 2022-01-14:
[server building](https://blog.csdn.net/qq_20451879/article/details/105272084)
there are main three methods for local sever building(broker), like Mosquitto, EMQTT.  As for the the downloading of emqtt is not so easy, so first tried mosquitto, maybe later try emqtt
here are several issues:
1. download mosquitto
2. go to mosquitto.exe to startup the server
3. open cmd and change terminal from c: to the file where mosquitto located in with terminal(here open two terminals, one is for subscriber, another is publisher ): _below is the publisher_ 
> C:\Users\hoshiraku>cd E:\c++_tencent_class\mqtt\mosquitto
> C:\Users\hoshiraku>e:
> E:\c++_tencent_class\mqtt\mosquitto>mosquitto_pub -d -t test -m this_is_jiabin

wjb321, 2022-01-14:
here is the new terminal of subscriber(this terminal should be opened first, because in this place it defines name: 'test' serve):
> C:\Users\hoshiraku>cd E:\c++_tencent_class\mqtt\mosquitto
> C:\Users\hoshiraku>e:
> E:\c++_tencent_class\mqtt\mosquitto>mosquitto_sub -d -v -t test
after another terminal publisher publish some info after -m in the terminal(space key is not allowed)
then subscriber will receive the messages from sub terminal
[image]=======publisher terminal 
[image]======subscriber terminal

wjb321, 2022-01-16:
there are two ways to run the demo of mqtt, either from the terminal of esp-idf 
>  idf.py  menuconfig         

or by using vscode esp-idf plugin interface

> wifi interface with uuid and password

then it works

wjb321, 2022-01-17:
can use two apps:
> mqtt.fx 

>tongxinmao

can start server either by mosquitto or emqx to start a  server. up part is the mosquito way. below is the emqx way to start:
>E:\c++_tencent_class\mqtt\emqx\bin>emqx start

then connect tongxinmao and mqtt.fx both to the wifi ip address
then both of them can publish and subscribe different topics and they can get info from each other.

wjb321, 2022-01-17:
try the esp32 but can not connect to the server, it needs to buy a cloud server and deploy the emqx or mosquito broker. then it can work.
otherwise it does not work, later i will check it out.

archerindigo, 2022-01-17:
The escape room has the server and broker set-up already so you just need to focus on the client stuff, and test with the server in university network.

wjb321, 2022-01-17:
cool, with the help of Terry, we are finally done with the testing of mqtt things. here i share some obstacles that i suffered.
1.  using the heavy but efficient iot framework from esp-idf, but stuck at the host configuration part(the server that i tested is not working anymore but i did not know that)
2. there are many debuggers(mqtt.fx, tongxinmao etc., need to know how to configure them.
3. there are several broker makers apps: mosquito and emqx(both are mentioned above) need to know how to start the server locally or remotely.
4. tried some clouds servers: emq x clods, ali clouds etc for using the servers
5. the debuggers might no info from publisher(tongxinmao), reopen it helps
[image]======basic configure in tongxinmao.app

.

wjb321, 2022-01-18:
now import the mqtt function into the BLE game, in order to test the function, set the win score to 64, and when reach the score, mqtt debugger can receive the payload info works properly.  here are some problems regarding c/c++:  the project is based on c++, and mqtt mainly based on c, and for the class which is from c++,  so there are plenty of drivers need be changed with 
> #ifdef __cplusplus  #endif

when try to encapsulate the mqtt  but there are some errors: multi redefinition of "XXX", which are mainly the classes, so stop that. finally works ok.
[image]
======step 1， start the local server
[image]
======step 2.    configure mqtt.fx and serial debugger with local server ip(my computer as the local server with an opening wifi)
[image]
======step 3,   test the result of mqtt if win the game

wjb321, 2022-01-18:
still there are some problems: game interface some backgrounds lost and not smooth(maybe delay problem etc.)
regarding the tasks mentioned above, it should be the last one, can be extended like this:
- [ ] when finishing game, mqtt sends messages, game quit with a pic showing on 
- [ ] game loss will a mqtt message and game restart 
- [ ] ......

wjb321, 2022-01-19:
after annotating the mqtt part, the game works fine. so i think the game blackout caused by mqtt, i will check out which part influence the game in the mqtt.

wjb321, 2022-01-19:
after checking the code step by step, when start:
> WiFi.begin(SSID, PWD); 

in mqtt, the game will have some blackouts, when stops it, the game works fine. i guess it is the channel conflict between Ble and wifi.

wjb321, 2022-01-19:
There are some updates when try json on the game:
[ArduinoJson 5 to 6 update ](https://arduinojson.org/v6/doc/upgrade/)

wjb321, 2022-01-19:
upper link is the official document, but the last part has little misleading codes, here should be the fine version for json mqtt: [Example json mqtt sending ](http://www.steves-internet-guide.com/arduino-sending-receiving-json-mqtt/)

wjb321, 2022-01-20:
- [ ] fix the bug of program crashing when typing command up and down or in in the homepage
- [ ] add aduino 6 json command format
- [ ] add game wining page

wjb321, 2022-01-20:
[image]
======final status

wjb321, 2022-01-20:
for the game crash problem: the code is same but works fine in only BLE status. 
the blackout is： 
1. new born number after moving, the original location will blackout 
2. if two numbers add up, the original locations will blackout, and next moving action will make the blackout fill up with zero, and new original location will blackout again(like 1)
3. 8 4 4, after moving left it wont be 16 but 8 8, i need send left command again(2 times) then it will be 16

wjb321, 2022-01-21:
Finally figure out the BLE and MQTT somehow signal interference, so now the project only use MQTT. by publish and subscribe the topics, now can control the game and send commands to mqtt.fx. and now there is no blackout in the game anymore.

wjb321, 2022-01-22:
solve the bugs when using mqtt only.
the bug is exit game when playing, the exit image(game interface)does not show up when quitting the game. it means the game only can go one direction and can not restart.

debug for a long time and find the problem is:
> if (act_obj == app_scr)
> {lv_scr_load_anim(app_scr, LV_SCR_LOAD_ANIM_NONE, 300, 300, false);}
this animation function removes the original image interface.but it works fine in BLE, not in mqtt(strange!)

wjb321, 2022-01-22:
Update: After winning a game, you do not need to stay in the you win screen. 
At this point the game will be restarted at regular time to prepare for the next group of players.